Sinapse-Energia/AP-Sinapse #192: [RF][Mismatching] To understand how is working now the RF TX - RX and improve matching with End Devices

ralcaide, 2018-12-21:
We have now three kind of End Nodes and not all of them are answering to the AP. However, they are answering to a legacy transparent modem.

| Gateway | EC Legacy | EC-NEMA | EC-RF |
|-------------|---------------|---------------|------------|
| Legacy  | OK | OK  | OK |
| AP | OK | KO | KO |

Normally, the new ECs execute the actions but they are not answering/sending the expected RF Frame.

We observed also that EC Legacy answers much more faster when the query comes from Legacy gateway (300ms) that from the query comes from AP (1000ms).

Next actions are required:

- Explanation about the RF sending speed from AP (All elements involved)
- Study mismatching and adapt speed in order to work well in every case

ralcaide, 2018-12-21:
@fpgamcu , could you please take a look to this task with very high priority, please?. We need at least an answer to the first two points asap. 
We will send you also all the devices in order to characterize in the best possible way what is happening but now we need at least increase the sending speed in order to perform empirical tests

ralcaide, 2018-12-21:
Additional questions:
1. How much time needs the function sendData (RF_Transceiver.c line 2071) for one PING frame?
2. How much time needs the function TDA5340_start_TX_FIFO (RF_Transceiver line 1660) for one PING frame?

ralcaide, 2018-12-21:
3. The function SPI_sendByte (RF_Transceiver.c line 1879) used HAL_SPI with 1000 as a timeout, in other versions of this driver uses 100 as a timeout. Could this fact provokes slower speed?

ralcaide, 2018-12-21:
Attached results file (resume in the description tables)

[RF_TEST.pdf](https://github.com/Sinapse-Energia/AP-Sinapse/files/2702675/RF_TEST.pdf)

fpgamcu, 2018-12-21:
3. In SPI_sendByte those 1000 timeout is only for compatibility with HAL calls. Whenever you use 1000 or 100 ticks it still will send data by SPI with SPI clock speed (652343 bit per second with current settings, prescaler 64, sys clock 167 MHz, bus clock 41.75 MHz)

fpgamcu, 2018-12-21:
1 and 2 - I am measuring now the time

fpgamcu, 2018-12-21:
1. Time spent on sendData (RF_Transceiver.c line 2071) is 146 msec
2. Time spent on TDA5340_start_TX_FIFO (RF_Transceiver line 1660) 131 msec

ralcaide, 2018-12-21:
a. So, 131 msec for 11 bytes (ping) means around 12 msec per byte, right? 
b. We suppose that TDA send the bytes at the same speed, is that right? or depends of something else?
c. Could we increase this speed?

ralcaide, 2018-12-26:
The code used by the end nodes not answering well to the AP is here:

https://github.com/Sinapse-Energia/EC-LC-NEMA/tree/fixstability

ralcaide, 2018-12-27:
I've created a file in order understand how is working each device. @fpgamcu , please fill the AP row and try to fill the other rows looking at the code of the other devices.

Here is the file https://docs.google.com/spreadsheets/d/1mUV6gDzYZ3d6Nbbc7lMgAZmTN7cqS2aGVIz_Ns6I4Cs/edit?usp=sharing

fpgamcu, 2019-01-04:
Status update:
Debugged with latest AP code and EC-LC-NEMA
Reproduced the issue: packets sending from AP to Node, Node receives packet, AP not receiving packets from node
Now checking RF transfer with analyzer, checking with debugged Node sending, checking with debugger AP receiving anything/broken packets

soporteHW, 2019-01-21:
Results from last test performed included in the following table:

[image]

ralcaide, 2019-03-12:
To be tested after the command 220; is done

closed 2019-07-09 by ralcaide