Sinapse-Energia/AP-Sinapse #193: [Config][MODBUS] Command to set the MODBUS configuration

ralcaide, 2019-01-23:
The MODBUS configuration could be modified in runtime. The command to set the MODBUS should contains:

- Parity
- Speed
- Number of bits

TODO RAE: Define command

ralcaide, 2019-01-23:
@fpgamcu , do you think other parameters should be modifiable in runtime?

after your feedback I will define the new command

fpgamcu, 2019-01-24:
130; modified to be able to set modbus uart parameters
Example 
130;2;9600;9;1;2; for 9600 baud 8E1 - 8 bits of data, even parity, one stop
130;2;115200;9;1;1; for 115200 baud 8O1 - 8 bits, odd parity, one stop
130;2;9600;8;1;0; for 9600 baud, 8N1 - 8 bits, no parity, one stop
Be aware that it is not saved into NVM by now and modbus uart parameters will be reset to default after restart.
Also according to ST uart parameters, when you set parity to not NONE, in order to have 8 data bits you should set data bits to 9

ralcaide, 2019-02-07:
The configuration of the UART / Modbus should be saved in NVM. Could you please provide an estimation about the time / implications to update the behaviour?

fpgamcu, 2019-02-13:
Should it be in SD card only or in SD and internal flash, to have possibility to work without SD?

ralcaide, 2019-02-13:
It depend. Saving them in flash will requires bootloader modification or not?
If yes: Only in SD
if not: Both, SD and internal flash