Sinapse-Energia/AP-Sinapse #195: [RF][Setup][220;] To setup the RF driver configuration in runtime

ralcaide, 2019-02-18:
Since [](https://github.com/Sinapse-Energia/AP-Sinapse/commit/f5de78c247f3b348d6a807c7044c65ca60dfcf4a) we are using a new driver but this one better working with EC-LC-NEMA than with EC-EVO. 
It is necessary to develop a MQTT command to select the driver and configure it in runtime in the AP and EN contexts. 

Commands to be developed
[220; - Action](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#customize-rf-driver)
[620; - Response](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#customize-rf-driver-result)

These values should be saved in the NVM 

Task related with #192

ralcaide, 2019-02-18:
@fpgamcu , please take a look to the task and tell us if it is clear. Then provide an estimation and do it with normal-high priority, please.

fpgamcu, 2019-03-05:
Runtime driver change added. But for now burst parameters and transparent mode is not changing in runtime, they are deeply in the defines and need to be converted into runtime variables. Now I am working on it.
One question: how should transparent mode work when we set it in runtime with 220 command? How is it interacting with 1; - 4; commands?

ralcaide, 2019-03-11:
As is working now. If the EPD is setup in the memory, then the 1; and 4; will work as expected. If the EPD is not setup in the memory, then 1; and 4; will not work (returning always 0). Is actually working like that?

ralcaide, 2019-04-02:
Regarding the delay we are already having in this ML this task has less priority than #201, #207 and #181

ralcaide, 2019-04-02:
@fpgamcu , status? How much time do you need to finish this task?

ralcaide, 2019-04-12:
@fpgamcu , status? We continue with big delay