Sinapse-Energia/AP-Sinapse #197: [Security] Private data should be encrypted before save in SD

ralcaide, 2019-02-25:
Private data is saved in plain text in the SD card. If someone get the SD card can compromise the whole system.
It is necessary to encrypt sensible data like:

- Broker (MURI, BMURI)
- User( MUSER, BMUSER)
- Password (MPSWD, BMPSWD)
- Update server (BTLSER and BTLNAM)

Another possibility is to save these data in MPU instead of SD but in order to do not create incompatibilities with BL we think is better to encrypt the sensible data saved in database.

An encryption library and valid mechanism should be chosen.  The key or certificate to encrypt should be saved in the MPU. This key or certificate will be used also for secure connection (SSL) to be developed in other task

ralcaide, 2019-02-25:
@fpgamcu , could you provide feedback - estimation about different possibilities?

fpgamcu, 2019-02-26:
ok, I'll check possible encryption schemes and write here the estimate.
is SSL certificate preferable as encryption key (using AES for encryption)?

ralcaide, 2019-03-08:
We should have the possibility to have both. But normally will be a SSL certificate. Could you provide feedback about estimation / complexity regarding different methods of encryption?

ralcaide, 2019-03-08:
After speak with costumer:

- To use SSL certificate that should be saved in MPU in manufacturing process
- To save the sensible data encrypted with SSL certificate in the MPU
- To use a secure cipher suite. Options? 
- Where is executed the cipher algorithm??
- It is necessary to implement MQTT over TLS (already done in EC-IoT and to be integrated)

@fpgamcu , could you provide feedback about previous items?

ralcaide, 2019-03-08:
Possible library to be used:
https://www.st.com/content/ccc/resource/technical/document/user_manual/34/1a/20/75/7f/84/45/cb/CD00208802.pdf/files/CD00208802.pdf/jcr:content/translations/en.CD00208802.pdf

ralcaide, 2019-03-12:
Updated description with technical requirements of the data signing. As we saw, ST has libraries to perform these encryptions and we would like to use them.
@fpgamcu , could you take a look to the requirements & library and provide us feedback about complexity & timeframe?

fpgamcu, 2019-03-13:
Info from @vAnArhist 

1. Protection data on flash.
MPU Memory protection unit is used for:
• Prohibiting the user applications from corrupting data used by critical tasks (such as the operating system kernel). 
• Defining the SRAM memory region as a non-executable (eXecute Never XN) to prevent code injection attacks.
• Changing the memory access attributes.
All this is runtime protection and is not exactly encrypt or protect flash from reading and rewriting.
MPU isn't suitable well for read-out protection. For this purpose is used Proprietary Code Read Out Protection (PCROP). The PCROP is a read and write protection of an IP-Code in Flash memory applied by sector to protect proprietary code from possible modifications or read out by the end user code, debugger tools or RAM Trojan codes. View more at https://www.st.com/content/ccc/resource/technical/document/application_note/89/12/c5/e2/0d/0e/45/7f/DM00186528.pdf/files/DM00186528.pdf/jcr:content/translations/en.DM00186528.pdf and https://www.st.com/en/embedded-software/x-cube-pcrop.html .
Issue here is that SSL certificates have limited lifetime and should not be one time writable, because each several years/months you will need to update certificate/key.

2. Encryption of data on SD-card.
STMicroelectronics provides crypto libraries: https://www.st.com/content/ccc/resource/technical/document/user_manual/group0/f9/6e/f2/a2/b4/ec/49/c0/DM00215061/files/DM00215061.pdf/jcr:content/translations/en.DM00215061.pdf . 
It is possible to use same cipher suite that provided with TLS certificate. In most cases it uses RSA256 private keys, so it could be used for public key in SD files encryption. Private key could be generated by hardware RNG.

Now @vAnArhist is working on encryption data on SD-card and internal flash using mbedTLS.
Sent yesterday request to ST to get native crypto libraries code, answer should in 72 hours.

ralcaide, 2019-03-13:
Thanks for the information. According to the customer requirements we will need:

- AES-256 with using GCM or CMC. 
Which method is more cost-effective? Please use the cost-effective one. 

Regarding crypto libraries code, is it not available?

Regarding keys and/or certificates: It is not clear for us how we will be generated or saved the keys/certificates. Could you propose a methodology?

vAnArhist, 2019-03-14:
@ralcaide 
1. I think we will use AES-256 with Galois/Counter Mode, it is more performance.
2. Crypto library from ST is only on request, as in some countries, not all encryption is legal due to licensing. I have now access to download this lib.
3. Yesterday I was checking the availability to perform asymmetric encoding (by RSA ciphers (public and private), which provides with TLS certificate, but it is not working with mbedTLS. 
4. I think that now it will be good to use hardware RNG to generate a cipher for AES-256.

vAnArhist, 2019-03-14:
@ralcaide  Some remarks:
Keys for certificates can be generated by RNG (needs to be checked) to take maximum entropy. The RNG is able to generate a 32‐bit random number at freq AHB/54. As we need 256-bit cipher we need to take 32-bits 8 times.
Cipher will be generated only once, then saved in flash. As we want to protect our key from reading we can use PCROP, as was decribed.
Please notice, that AES can process only data blocks of 128 bit.
One of way to generate the key by openSSL: https://security.stackexchange.com/questions/126988/ways-to-generate-symmetric-and-asymmetric-keys

ralcaide, 2019-03-25:
@vAnArhist , could you please move all the previous commits to a new branch Commboard-V2.0-F427VG_SECU and remove them from the current branch? 
We are making some very delicate changes in the current branch and we don't want to have any risk. We will test all the security features in another branch and then integrate all of them into the current one.

fpgamcu, 2019-03-25:
We will do it together with @vAnArhist

ralcaide, 2019-05-06:
@fpgamcu , @vAnArhist what is the status of this??

fpgamcu, 2019-05-06:
Status:
Code that is now pushed is encrypting SD card files using AES-256
Key is stored inside shared memory
There is command to change key, SET_RSA_KEY;newkey; but after change key SD card internal data is lost
TODO:
add code to reencrypt SD card with new key when changing it, to not lost the data
cleanup last commit (there is some not necessary lines added/removed)
flash protection in shared flash - this aspect needs additional discuss
Ivan had checked on flash locking, but we need to work on it more, because it can make flash nnot accessible by bootloader/jtag

ralcaide, 2019-05-08:
We need at least the private data encrypted in the SD, even with a Hardcoded key. When will be able to test that? The available code is already testable?

ralcaide, 2019-05-14:
@fpgamcu , could you please answer my questions? Are we able to test that?

fpgamcu, 2019-05-14:
Yes, latest code should work. Only thing that latest commit, where key was moved to shared memory, is not compatible with current bootloader

ralcaide, 2019-05-14:
So what bootloader should we use?

fpgamcu, 2019-05-14:
You can use any version, but shared data should be reset to default by client and bootloader whenever it starts. Or I can modify current code to not use BL by now.

ralcaide, 2019-05-14:
no, no problem. We can use the by default key

ralcaide, 2019-06-06:
We are testing with bootloader 1.46 and it is not working. Here is the diff. Should we leave the RSA_KEY empty?
[secu_diff.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/3260228/secu_diff.txt)

ralcaide, 2019-06-06:
@fpgamcu , @vAnArhist , are you able to create a new branch in the bootloader and perform the required modifications?

vAnArhist, 2019-06-06:
@ralcaide  Hello, it should work on bootloader without modifications. Ok, we will check and add modifications if required.

ralcaide, 2019-06-06:
The client with generated with the code displayed in the diff is not working. Should we set the RSA_KEY to empty?

vAnArhist, 2019-06-06:
need to check the code to answer

ralcaide, 2019-06-11:
First testing done with following results:

- BL + CL connects OK
- SD content is encrypted
- Change broker with 5; works OK
- After restart with 398; the device connects to the previous broker. This is a BUG because the device should connect to the last valid broker.

@vAnArhist , @fpgamcu , could you please take a look to this bug and fix it asap?

ralcaide, 2019-06-17:
@fpgamcu , @vAnArhist, could you please provide feedback? We need this fix in order to send the devices to some customers requiring encryption.

fpgamcu, 2019-06-17:
We will test it today

fpgamcu, 2019-06-17:
Tested just now with next flow:
1) On start connected to soporte-tecnico.bitnamiapp.com : 1883
2) Send `5;fw.x5r7.com;1883;;;` to connect to fw.x5r7.com : 1883
3) Connected to fw.x5r7.com : 1883
4) Reboot with `398;0;`
5) After reboot connected to fw.x5r7.com : 1883
6) Power off reboot
7) After reboot connected to fw.x5r7.com : 1883

Working ok.

Few things to notice:
1) According to our code, device parameters saved into SD only once a minute or on 398;0; so if you power off reboot right after change broker, it may not work.
2) If you use incorrect broker, it may return to previous one.

ralcaide, 2019-06-17:
Thank you, we think the problem is related with 398 speed. We will test and let you know