Sinapse-Energia/AP-Sinapse #201: [API][Change APN - 175; ] To implement a command to change de APN

ralcaide, 2019-03-12:
TODO by RAE

ralcaide, 2019-03-12:
@fpgamcu , please read the requirements and provide feedback about estimation & feasibility.

fpgamcu, 2019-03-18:
Maximum estimated time to implement and test is 16 hours.

fpgamcu, 2019-04-02:
Please test when possible.
One small remark: APN + USER + PASSWORD should be lesser than SIZE_APN - 10. By default SIZE_APN is 60. And APN, USER, PASSWORD each should be lesser than 64 symbols.

ralcaide, 2019-04-03:
Thank you, we will test asap.
I don't understand very well the comment. Where is defined SIZE_APN? If by default SIZE_APN is 60, then APN + USER + PASSWORD should be max 50 characters? And why only APN could be 64 characters size? 
Please, explain us better how the maximum sizes works

fpgamcu, 2019-04-03:
SIZE_APN is defined in Definitions of BL and CL
It is 60 characters for APN string buffer in M95 driver. M95 driver using this buffer for next string:
"APN","USER","PASSWORD"\r
This " and ',' and \r symbols is taking 9 chars from buffer, and null terminator in the end of the string is taking also one char. So we have 50 chars for APN, USER and PASSWORD total. This is important because inn M95 driver there is no check for this three variables is less than 50 characters in total.

About 64 - it is size for each variable in shared memory, so if you specify for example USER more that 64 chars, only 64 would be saved into shared memory.

This 2 limits is not linked, that is why it have no logical connection each other. I only wanted to remind that we have such limitation.

ralcaide, 2019-04-03:
But they are in fact linked, isn't? If APN+USER+PASSWORD is > 50 the M95 driver will not work so APN = 64 is not an option, isn't?

fpgamcu, 2019-04-03:
yes, I mean limitation isn't linked, but variables linked in any way. I this 64 limitation for single variable is not significant, since APN+USER+PASSWORD <= 50 covers it too.

ralcaide, 2019-04-03:
Ok, so the real limitation is APN+USER+PASSWORD <= 50. Could we detect in compilation time if this addition is > 50 and then avoid the generation of the FW?. I'm sure we will reach this limit of 50 in some cases.
Is it possible to increase this size easily?

fpgamcu, 2019-04-03:
Yes, change SIZE_APN in Definitions
You also need to change it in both CL and BL.

ralcaide, 2019-04-03:
OK. Please revert commits related with this task until #207 is fixed in order to do not have any risk updating the CMCs already installed in the field. 

Then, once the schedulers works as expected in CMC we will integrate the Change APN command.

ralcaide, 2019-04-16:
In progress: This development is now in an independent branch and should be integrated in the master. It requires a new bootloader

soporteHW, 2019-05-13:
Testing the new Binary, following the steps:

- Waiting for the device to connect to the broker
- Sended "175; TM ;;;" for TM SIM
- Answer "675; matooma.m2m ;;; TM ;;;"
- Restart with 398 (also tested with manual restart)
- After several attempts, do not take the TM card. It restarts and returns to the matooma configuration (with the matooma card inserted, if it is restarted, it connects)