Sinapse-Energia/AP-Sinapse #202: [207;][BUG] [BL related] If PATH is not empty or / the update is not working

ralcaide, 2019-03-12:
The BL is not downloading the correct file when the PATH is not empty. For example:

- OK: 607;843168;0;AP_STARTUP_1.46.1;www.eluxoon.com;80;HTTP;/;CL-F427VG-AP-sinapse-AP_GENERIC-IMEI-matooma.m2m-V1.46.1.bin;;;9; 

- KO : 607;843168;0;AP_STARTUP_1.46.1;www.eluxoon.com;80;HTTP;/ap_fw/;CL-F427VG-AP-sinapse-AP_GENERIC-IMEI-matooma.m2m-V1.46.1.bin;;;9;

- KO: 607;843168;0;AP_STARTUP_1.46.1;www.eluxoon.com;80;HTTP;ap_fw/;CL-F427VG-AP-sinapse-AP_GENERIC-IMEI-matooma.m2m-V1.46.1.bin;;;9;

soporteHW, 2019-03-13:
Failed Test: https://sinapseenergia.testrail.io/index.php?/tests/view/1526&group_by=cases:section_id&group_order=asc&group_id=177

If you enter data in the path, it does not take the data.

ralcaide, 2019-03-14:
We've found a workaround: To use the path directly in the filename and the path empty. 

It is clear that the bootloader does not use the path, only the filename. In fact, this is not a problem. We will use the path+filename as filename. Anyway we found a limitation in the size of the filename. It seems with big strings is not working. For example, this one: `;ap_fw/development/sinapse/1461/CL-F427VG-AP-sinapse-AP_GENERIC-IMEI-matooma.m2m-V1.46.1.bin`

@fpgamcu , could you tell us what is the maximum size for the filename?. Is it possible to increase it easily?

fpgamcu, 2019-03-18:
Yes, indeed, bootloader code is not using path, only filename field.
Limitation in bootloader code is next:
HTTP GET request should be not more than BOOT_BUFFER_SIZE (Bootloader.h) 4096 bytes by default
this request is made up "GET /%s HTTP/1.1\r\nHost: %s\r\n\r\n" so host name + file name/path should have not more than 4070 bytes.
But host name and file name also limited by parameters sharing lib.
In Shared.h in bootloader and client host name is limited by 64 bytes and file name also by 64 bytes.
It could be easily changed, byt need to be changed in both BL and CL.

ralcaide, 2019-03-18:
Thanks, by the moment we will continue working without modifification, knowing that the limit is 64 bytes for the filename.

ralcaide, 2019-03-18:
One more question: Is there a workaround modifying only the client? We don't want to modify the BL if it is not necessary

fpgamcu, 2019-03-18:
Unfortunately no, because bootloader is managing connection, we need to modify it to handle other length.