Sinapse-Energia/AP-Sinapse #203: [CMC - Metering][EASTRON]  To display imported energy instead exported

ralcaide, 2019-03-15:
The L3_READ_METERING_* commands are displaying the exported energies instead the imported. 

[EASTRON]: Please change that in order to display the imported energies
[CIRCUTOR]: To verify what energy is got, it is necessary also to display the imported energy.

Example of diff is attached

[this_diff_-_.c.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/2971272/this_diff_-_.c.txt)

fpgamcu, 2019-03-18:
Estimation is 2 hours, 
Have you external PC with circutor analyzer connected?

ralcaide, 2019-03-18:
Not at this moment. We will do that and I will let you know.
We tested in Eastron and in this case is necessary to change the Exported for Imported.

soporteHW, 2019-04-23:
Tested: https://sinapseenergia.testrail.io/index.php?/tests/view/1629&group_by=cases:section_id&group_order=asc&group_id=170

OK

[image]

ralcaide, 2019-04-25:
Great, we should test also with a the Circutor analyzer in order to check of we are using the correct energy or not. 
@soporteHW , could you do that, please?

closed 2019-07-09 by ralcaide