Sinapse-Energia/AP-Sinapse #204: [CMC - Schedulers][BUG] Bad behaviour of CMC after restart setting relays to wrong status

ralcaide, 2019-03-21:
In a real project we are facing a very problematic bad behaviour:

- Versions of the FW: BL 1.46 & CL 1.46.1
- Device setup from scratch and start with cmc relays = 0 - > OK
- A scheduler based in SS/SR is sent: 10;REL10;SR-10;SS+15;-1;
- At 23:01 UTC the device is restarted unexpectedly. At this moment, the relay1 = 1.
- When the device connect again the relay1 = 0 -> KO . Very bad behaviour because all the lights are OFF during the night.
- The same scheduler in SS/SR is sent again and then the relay1 = 1 again.

We are not able to reproduce this problem in another device in Sinapse. But it is happening in a real field device.

Analyzing the behaviour it seems there are two problems:
- The CMC does not maintain the relay status after restart 
- The CMC does not maintain the scheduler after restart

Differences with the device in Sinapse:
- The real field device have a lot of EPDs close to it, so a lot of RF messages are in the air.
- The real field device have received wrong commands in some moments, commands like: 10;REL10;SR-10;SS15;-1; and 10;REL10;SR-10;SS+15; . After these commands the device was updated with 314; and 208; so the client was update. Also, the correct scheduler was sent.

@fpgamcu , any idea? Are you able to reproduce this behaviour? Is maybe related with RF or with the curruption of SD or Flash?

fpgamcu, 2019-03-21:
I also was not able to reproduce the issue yesterday, but have set up not a lot of EPDs, now trying to set up as much as possible EPDs and reproduce.

Saving relays and schedulers in internal flash works next way: when CL starts, it read from internal flash relays and schedulers with checksum. If checksum is bad - it sets relays to 0 and schedulers to not running. same check is also made by BL when it starts.
After CL read relays and schedulers, it saves it in the engine, and then uses it values from the engine.
Each minute and when restart command triggered CL saves relays and schedulers from engine to internal flash.

I see next possibilities:
1) Engine content was corrupted by malformed command and relays values was set to 0, and then saved into the internal flash when rebooting.
2) Internal flash was somehow corrupted by CL before reboot or bootloader after flashing new CL, and check for checksum fails, erasing corrupted internal flash sector.

Question: what is the size of CL binary used?

ralcaide, 2019-03-21:
The size of the CL is: 201832 Bytes.
It seems the flash is corrupted and the checksum fails, yes. It is important to know that 314; and 208; was sent after the use of corrupted commands. Also, please note that the 10; is sent after each new connection to the CMC, so should be saved in the Context and in the Flash and then the flash should not be corrupted. Something that we don't understand is happening

fpgamcu, 2019-03-21:
This size should be ok.
If it is flash corrupting it is very bad, because in code we exactly don't have a place where flash could be corrupted.
Much stuff is protected by checks also.
But, for example, in Bootloader.c in Boot_PerformFirmwareUpdate() we exactly don't have any protection by firmware size.
Also another possible issue is in this part of code:
```c
for (i = 0; i < fw_len; i += 128)
{
    FlashNVM_Read(fl_addr + i, (uint8_t*)boot_buff, BOOT_BUFFER_SIZE);
    FlashNVM_Write(app_addr + i, (uint8_t*)boot_buff, BOOT_BUFFER_SIZE);
    if (WDT_ENABLED == 1) HAL_IWDG_Refresh(&hiwdg);
}
```
We are reading and writing BOOT_BUFFER_SIZE but moving from read to read only by 128 bytes, this is working only because we rewriting same data and flash write is successful, but still it is an issue, because BOOT_BUFFER_SIZE is 4K we are writing same data about 16 times, not an issue, but flash will be aging 16 times faster.
According to this datasheet https://www.st.com/resource/en/datasheet/stm32f427vg.pdf Table 50
Flash endurance is 10000 cycles. 10000/16 - after 625 updates of CL firmware MCU flash will not guaranteed work.
Here I can see the issue about write cycles - our shared internal flash area could be updated only 10000 times, after this count we could expect flash corruptions. Each restart, each change of scheduler for relay or relay state is one write into this shared flash, so we can calculate device lifetime based on it.

ralcaide, 2019-03-21:
Ok, anyway, this device is new, supposing the flash is still working, we need to find a reason of the bad behaviour explained in the description and try to avoid it.
As you say in the #204, if the flash is corrupted, then is reset, right? So, we suppose the flash is OK and is telling that the relay is 0?

ralcaide, 2019-03-22:
Problem found: Use of a CUR_FW_VERSION > 8 bytes. It will be tested and the issue will be closed if OK

soporteHW, 2019-04-17:
Tested OK

Relay1: https://sinapseenergia.testrail.io/index.php?/tests/view/1578&group_by=cases:section_id&group_order=asc&group_id=130

Relay 2-7: https://sinapseenergia.testrail.io/index.php?/tests/view/1613&group_by=cases:section_id&group_order=asc&group_id=130

closed 2019-07-09 by ralcaide