Sinapse-Energia/AP-Sinapse #205: [BL][207;][BUG] CMC starts with wrong relay status if the filename to search is too big

ralcaide, 2019-03-21:
Observed behaviour:

- BL 1.46.0 and CL 1.46.1
- BL hardcoded the filename: "ap_fw/development/ap_generic_gprs.bin"
- Device from scratch
- After download the client the first time all the relays are set to 1 -> KO 

Then

- BL 1.46.0 and CL 1.46.1
- BL hardcoded the filename: "ap_fw/startup/ap_startup_gprs.bin" (Same client as previous case but other name)
- Device from scratch
- After download the client the first time all the relays are set to 0 -> OK

The only difference between one test and other one is the filename. The FW is the same. 

It seems there is a bad management of the filename and the shared flash.

ralcaide, 2019-03-21:
Related with: #204 and #202

ralcaide, 2019-03-21:
@fpgamcu , it seems there are problems with the shared flash, isn't??. We need to secure the management and the use of the shared flash. Could you provide us the flash memory map and how are saved there the variables?

fpgamcu, 2019-03-21:
Can you please share part of bootloader code where you hardcode the filename?

ralcaide, 2019-03-21:
sure 
`#ifndef _DEFINITIONS_H_
#define _DEFINITIONS_H_

#ifdef __cplusplus
 extern "C" {
#endif

// *************************************************************************************** //
// *************************************************************************************** //
// *************************************************************************************** //

 // RAE definitions
#define COMMUNICATION_M95
#define HYBRID_M95_WLAN_CODE

 // Francis definition.
#define CMC_APPLICATION_DEPENDENT
#define LOG_WIFI  0
#define NUMBER_RETRIES 2
 /// Dependent Application
#define START_EEPROM_OWN_CMC_PARAMETERS 	0x10
 /// direcciones dispositivos I2C
#define I2C__ADDRESS__READING__DS1338    0xD1
#define I2C__ADDRESS__WRITING__DS1338    0xD0
#define I2C__ADDRESS__READING__AT24C512  0xA9
#define I2C__ADDRESS__WRITING__AT24C512  0xA8

#define CIRC_RELEGENERAL_Pin GPIO_PIN_6
#define CIRC_RELEGENERAL_GPIO_Port GPIOE
#define CIRC_RELE1_Pin GPIO_PIN_5
#define CIRC_RELE1_GPIO_Port GPIOE
#define CIRC_RELE2_Pin GPIO_PIN_4
#define CIRC_RELE2_GPIO_Port GPIOE

#define CIRC_RELE3_Pin GPIO_PIN_3
#define CIRC_RELE3_GPIO_Port GPIOE
#define CIRC_RELE4_Pin GPIO_PIN_2
#define CIRC_RELE4_GPIO_Port GPIOE
#define CIRC_RELE5_Pin GPIO_PIN_1
#define CIRC_RELE5_GPIO_Port GPIOE
 // *************************************************************************************** //
 // *************************************************************************************** //
 // *************************************************************************************** //

 // GPRS init
#define SIZE_APN 						60
//#define SIZE_MAIN_SERVER 				60
//#define SIZE_NTP_SERVER 				60
#define SIZE_GPRS_BUFFER 				256
#define SIZE_WIFI_BUFFER 				(4*1024lu)//256
#define TIMING_TIMEOUT_GPRS 			20
//#define const_APN 						"\"im2m.matooma.com\",\"movistar\",\"movistar\"\r\0"
//#define const_APN 						"\"m2m.tele2.com\",\"tele2\",\"tele2\"\r\0"
#define const_APN 						"\"matooma.m2m\",\"\",\"\"\0"
//#define const_APN 						"\"orangeworld\",\"orange\",\"orange\"\r\0"
#define const_SERVER_NTP 				"\"0.europe.pool.ntp.org\"\r\0"
//#define const_MAIN_SERVER				"\"m2m.eclipse.org\",1883\r\0"
//#define const_MAIN_SERVER 				"\"178.94.164.124\",80\r\0"
//#define const_MAIN_SERVER 				"\"89.248.100.11\",80\r\0"
//#define const_MAIN_SERVER 				"\"sinapseenergia.com\",80\r\0"

 // Wait for HTTP server answer (sec)
#define TIMING_TIMEOUT_UART 			15 // This should be lesser than 30 in order to keep watchdog
// HTTP connection
//#define HTTP_SERVER_IP 					"178.94.164.124"
//#define HTTP_SERVER_IP 					"89.248.100.11"
#define HTTP_SERVER_IP 					"www.eluxoon.com"
/*
#define HTTP_SERVER_IP_03				178
#define HTTP_SERVER_IP_02				94
#define HTTP_SERVER_IP_01				164
#define HTTP_SERVER_IP_00				124
 */
#define HTTP_SERVER_IP_03				89//178
#define HTTP_SERVER_IP_02				248//94
#define HTTP_SERVER_IP_01				100//164
#define HTTP_SERVER_IP_00				11//124

#define HTTP_SERVER_PORT				80
//#define HTTP_SERVER_FW_FILENAME		"firmware_crc.bin"
//#define HTTP_SERVER_FW_FILENAME		"wifi_080617.bin"
//#define HTTP_SERVER_FW_FILENAME		"hello.bin"
//#define HTTP_SERVER_FW_FILENAME		"hello_8000.bin"
//#define HTTP_SERVER_FW_FILENAME			"led_p1_8000.bin"
//#define HTTP_SERVER_FW_FILENAME		"myfile.bin"
 //#define HTTP_SERVER_FW_FILENAME		"CMC.bin"
#define HTTP_SERVER_FW_FILENAME		"ap_fw/startup/ap_startup_gprs.bin"

 // Define Bank Sectors (11 sectors total - for STM32F4xx devices)
 // Bootloader
#ifdef STM32F4
 #define FLASH_BANKB_START_SECTOR		FLASH_SECTOR_0
 #define FLASH_BANKB_SECTORS			2
 // Application
 #define FLASH_BANKA_START_SECTOR		FLASH_SECTOR_2
 #define FLASH_BANKA_SECTORS			6
 // Application copy
 #define FLASH_BANKC_START_SECTOR		FLASH_SECTOR_8
 #define FLASH_BANKC_SECTORS			3
 // Shared memory address
 #define FLASH_BANKD_START_SECTOR		FLASH_SECTOR_11
 #define FLASH_BANKD_SECTORS			1

#else

 // Define Bank Sectors (16 sectors total 4kb/each - for STM32F0xx devices)
 // Bootloader
 #define FLASH_BANKB_START_SECTOR		FLASH_SECTOR_0
 #define FLASH_BANKB_SECTORS			2
 // Application
 #define FLASH_BANKA_START_SECTOR		FLASH_SECTOR_2
 #define FLASH_BANKA_SECTORS			6
 // Application copy
 #define FLASH_BANKC_START_SECTOR		FLASH_SECTOR_8
 #define FLASH_BANKC_SECTORS			4
#endif

// App version addressing (+ release data address dd/mm/yy)
// placed in first address of the Firmware/Application memory
#define APP_VER_ADDR_LOW			0x0190
#define APP_VER_ADDR_HIGH			0x0191
#define APP_VER_ADDR_DD				0x0192
#define APP_VER_ADDR_MM				0x0193
#define APP_VER_ADDR_YY				0x0194

// WIFI ETH Mode
#define TRANSPARENT_WLAN

#define PERFORM_WLAN_FIRST_TIME_CONFIG

// Server IP should not be in the same subnet with client IP
#define WLAN_SERVER_CONFIG "AT+LANN=10.10.100.254,255.255.255.0"

// Define Client configuration for Ethernet and STA WiFi
//#define WLAN_CLIENT_CONFIG "AT+WANN=static,192.168.1.97,255.255.255.0,192.168.1.1"
#define WLAN_CLIENT_CONFIG "AT+WANN=DHCP,192.168.1.97,255.255.255.0,192.168.1.1"

// Define time need to reboot and establish connection in msec
// 15 sec for Ethernet and 30 sec for WiFi (connect to AP, etc.)
// TODO change this delay to waiting for some AT command checking connection
//#define WLAN_CONNECT_TIME	15000
#define WLAN_CONNECT_TIME	30000

#ifdef __cplusplus
}
#endif

#endif`

fpgamcu, 2019-03-21:
Flash map:

Bootloader:       BANKB 0x08000000-0x08007FFF 32K max
Application:      BANKA 0x08008000-0x0807FFFF 512K max
Application copy: BANKC 0x08080000-0x080DFFFF 384K max
Shared memory:    BANKD 0x080E0000-0x080FFFFF 128K max

ralcaide, 2019-03-21:
Thanks.

I see something very problematic. The BL that we are using uses 85K!!

-rw-r--r-- 1 1000 1000 84571 Mär 12 10:18 BL-F427VG-eluxoon-matooma.m2m-WIFI-ap_startup_wifi-V1460.hex

How is that possible? Why is the device working??

fpgamcu, 2019-03-21:
hex files size is not the flash size occupied by the firmware, because hex is text format and not binary
you need to convert it to binary to get occupied flash size, or use some tool to calculate flash size from hex

fpgamcu, 2019-03-21:
https://www.avrfreaks.net/forum/hex-file-size-flash-size

ralcaide, 2019-03-21:
Ok, thanks. The siye in binary is 30K:

-rwxr-xr-x 1 1000 1000 30492 Mär 21 12:05 bootloader-stm32.binary

So, we suppose the BL size is correct. 
What about the shared flash?. Are the variables there size limited?. Is possible to corrupt a variable because the big size of another one? . Is it possible to reset the shared flash?

fpgamcu, 2019-03-21:
We have size checks in Shared.c library. From first look we have checks in all places working with shared memory, but I am checking the code now, maybe something is missed. Should not be possible to corrupt one value because big size of another one. Also reset is done only when read from flash was failed (flash is broken or voltage is bad) or CRC error (flash is corrupted).

fpgamcu, 2019-03-21:
Can I try with my device with this two files? Is server up?

ralcaide, 2019-03-21:
Sure, this file will download a file that is at ap_fw/startup/ap_startup_dhcp.bin and will publish at AP_STARTUP in our broker. Anyway, please focusin #204. It is our biggest problem right now 

[BL-F427VG-eluxoon-matooma.m2m-DHCP-ap_startup_dhcp-V1460.hex.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/2991843/BL-F427VG-eluxoon-matooma.m2m-DHCP-ap_startup_dhcp-V1460.hex.txt)

ralcaide, 2019-03-22:
Problem found: Use of a CUR_FW_VERSION > 8 bytes. It will be tested and the issue will be closed if OK

soporteHW, 2019-04-17:
Tested OK

Relay1: https://sinapseenergia.testrail.io/index.php?/tests/view/1578&group_by=cases:section_id&group_order=asc&group_id=130

Relay 2-7: https://sinapseenergia.testrail.io/index.php?/tests/view/1613&group_by=cases:section_id&group_order=asc&group_id=130

closed 2019-07-09 by ralcaide