Sinapse-Energia/AP-Sinapse #207: [CMC Schedulers][10;][BUG] CMC schedulers are lost after restart

ralcaide, 2019-03-25:
The CMC schedulers (10;) and its interaction with the on demand actions (11;) works good if the device is not restarted. After a restart the scheduler does not work anymore.

ralcaide, 2019-03-25:
Tested with SS/SR and fixed times.
@fpgamcu , this is very problematic bug because the device normally restarts to synchronize with NTP and then the scheduler is lost.

fpgamcu, 2019-04-01:
Tested with latest code and next sequence:
At 2:55 relays is off, executed
```10;REL71;03:00;03:02;-1;```
At 3:00 relay 7 was turn on
At 3:02 relay 7 was turn off
At 3:03 executed
```10;REL71;03:10;03:12;-1;```
```398;0;```
At 3:10 relay 7 was turn on
At 3:12 relay 7 was turn off

Relays schedulers working ok after restart.

Have you some specific conditions during the test?

ralcaide, 2019-04-01:
We've tested again and surprisingly, the relay 7 is working good and the relays 1 and 2 are not working as expected. @fpgamcu : Any idea? Could you please reproduce it and fix it?
Attached images displaying results:

Relay 1:

[image]

Relay 2:

[image]

Relay 7:

[image]

fpgamcu, 2019-04-03:
Can you remind me where is GPIO D3 (RELAY1) located on pin header?

francisjjp, 2019-04-03:
Pin number 9 in H_27_1 50 vertical header connector.

fpgamcu, 2019-04-03:
[image]
This one?

ralcaide, 2019-04-03:
@fpgamcu , this is not a CMC. This is a EasyNet. We don't understand the problem. Could you explain better, please?

fpgamcu, 2019-04-03:
We have no CMC with F427 here in office, only old ones.
I will try to test somehow relay1 with EasyNet.

fpgamcu, 2019-04-03:
I can test with commands only, not a problem.

francisjjp, 2019-04-03:
[image]

GPIO_PD3 in EN_SA_v3p0 is as shown in bottom layer over R_37_17

francisjjp, 2019-04-03:
What version of EN EasyNet do you have?

fpgamcu, 2019-04-03:
Thanks for image.
Not sure what EN I have, it is I think about year old, maybe 1.5 year

fpgamcu, 2019-04-03:
Just tested with Relay1

Sent ```11;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;FALSE;``` at 11:15
got ```L3_READ_RELAY_STATUS_R;E70C08;0;0;0;0;0;0;0;1554290160;```

Sent ```10;REL11;11:20;11:22;-1``` at 11:16
got ```L3_READ_RELAY_STATUS_R;E70C08;0;0;0;0;0;0;0;1554290280;```

Sent ```11;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;FALSE;``` at 11:18
got ```L3_READ_RELAY_STATUS_R;E70C08;0;0;0;0;0;0;0;1554290341;```

Sent ```11;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;FALSE;``` at 11:21
got ```L3_READ_RELAY_STATUS_R;E70C08;1;0;0;0;0;0;0;1554290400;```

Sent ```11;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;FALSE;``` at 11:23
got ```L3_READ_RELAY_STATUS_R;E70C08;0;0;0;0;0;0;0;1554290640;```

Sent ```10;REL11;11:30;11:32;-1``` at 11:25
got ```L3_READ_RELAY_STATUS_R;E70C08;0;0;0;0;0;0;0;1554290700;```

Sent ```398;0;``` at 11:26
got device restarted

Sent ```11;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;FALSE;``` at 11:27
got ```L3_READ_RELAY_STATUS_R;E70C08;0;0;0;0;0;0;0;1554290820;```

Sent ```11;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;FALSE;``` at 11:31
got ```L3_READ_RELAY_STATUS_R;E70C08;1;0;0;0;0;0;0;1554291062;```

Sent ```11;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;ACTX;FALSE;``` at 11:34
got ```L3_READ_RELAY_STATUS_R;E70C08;0;0;0;0;0;0;0;1554291240;```

As you can see, relay 1 scheduler is working after restart

But this with new bootloader
I will also test with old one

ralcaide, 2019-04-03:
Strange. We are testing with BL 1.46.0 and CL 1.46.1 . Gprs version. For us is the scheduler is lost after restart as you can see in our previous comments

ralcaide, 2019-04-03:
Any advancement here? Are you able to reproduce the problem?

capacitorsmith, 2019-04-03:
I was not able to test with GPRS. In about an hour I will run exact test BL 1.46.0 and CL 1.46.1. GPRS version.
Could you please provide diff of sources for CL and BL that you used to prepare binary compared to that we have in github. Any small change like APN or context define could give a clue.

fpgamcu, 2019-04-03:
Sorry, those was different github account.
I was not able to test with GPRS. In about an hour I will run exact test BL 1.46.0 and CL 1.46.1. GPRS version.
Could you please provide diff of sources for CL and BL that you used to prepare binary compared to that we have in github. Any small change like APN or context define could give a clue.

ralcaide, 2019-04-03:
Context: AP
APN: matooma.m2m
user / password: Empty
TOPIC_BASE: AP_STARTUP
FW_VERSION: AST1461

At this moment I don't have the diff but I will try to provide you it tomorrow. 
Please, let me know the results of your test

ralcaide, 2019-04-04:
Attached the diffs of BL (1.46) and CL (1.46.1)

[bl_diff.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/3042015/bl_diff.txt)
[ap_diff.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/3042016/ap_diff.txt)

ralcaide, 2019-04-04:
@fpgamcu , are you able to reproduce the problem?

fpgamcu, 2019-04-04:
Yes, was able to reproduce it with GPRS. Now debugging.

ralcaide, 2019-04-05:
How is going this task? Could we have a fix soon? If not, is there a workaround? This bug is creating us problems.

fpgamcu, 2019-04-08:
Found bug in CL V1.46 - relay schedulers are saved into NVM only if some relay change its state after setting a scheduler. This bug is already fixed in CL V19.8.0

ralcaide, 2019-04-08:
Ok. We will test it in new version

ralcaide, 2019-04-16:
Tested OK

Relay1: https://sinapseenergia.testrail.io/index.php?/tests/view/1578&group_by=cases:section_id&group_order=asc&group_id=130

Relay 2-7: https://sinapseenergia.testrail.io/index.php?/tests/view/1613&group_by=cases:section_id&group_order=asc&group_id=130

closed 2019-06-26 by ralcaide