Sinapse-Energia/AP-Sinapse #212: [6LoWPAN][ER-EN][BUG] ER listen EN status only after several restarts

ralcaide, 2019-04-09:
Observed bug:
1. Setup 3 devices
2. Act over devices (3;X;): OK
3. Ask status several time(1;): Everything is 0. 
4. Restart ER
5. Ask status (1;): One device answer with timestamp different to 0
6. Restart ER
7. Ask status(1;): Now answer to devices....

After several restart answer all the devices

ralcaide, 2019-04-09:
Also, after answer about the setup EPDs we saw there is corruption in the SD:

713;B375E8;47;;0;100181;35;140267;35;140279;35;140285;35;140371;35;140439;35;514853;35;520002;35;527836;35;529333;40;532941;35;770001;40;770001:40;0;770002;40;770002:40;0;770003;150;770003:40;0;770004;40;770005;40;770016;40;770017;40;770018;40;770019;40;770020;40;770021;40;770022;40;770023;40;770024;40;770025;30;770025:40;0;770026;35;770026:40;0;770027;35;770028;60;770029;40;77003;30;770030;30;770031;40;770031:40;0;770032;40;770033;40;770034;30;770034:40;0;770035;40;770062;30;999999;100;

ralcaide, 2019-04-09:
@fpgamcu , any idea??

ralcaide, 2019-04-10:
The corruption of SD is not happening normally. 
The communication with the EN is still not good. It is necessary to restart the ER several time in order to get communication from it.

ralcaide, 2019-05-02:
Testing without success. The workaround seems is not working. Test:

- Connect ER
- Connect 5 ENs
- Setup 5 ENs
- RALL;
- Wait 30s
- 398;
- Getting answer only from 2 ENs
- Restart several times the ER with 398; and reach communication with maximum 4 ENs.

@fpgamcu 
- Should we wait more time?
- Sadly it seems is not working very well. We need to find a solution to that asap. Have you already asked to ST?

ralcaide, 2019-05-02:
Tested again but waiting 5 minutes instead 30s and the result is almost the same:
Communication with only 2 devices, but these devices are different to the previous one

ralcaide, 2019-05-02:
In brief, it seems the behaviour is still the same. Random and not good.

fpgamcu, 2019-05-02:
Created ST question on forum https://community.st.com/s/question/0D50X0000Algj2HSQQ/having-trouble-with-xcubesubg1

I got better results while using 398;0; right after RALL;

ralcaide, 2019-05-02:
We tested again and we got again the same results: 2/5
How are you testing?
- Are you able to test the with ST evaluation boards and the original driver? 
- Is the problem reproducible also with the original driver?
- Is the sender maybe blocked by the receiver?

fpgamcu, 2019-05-03:
Same on st boards with original driver (there is not much changes in original driver comparing to ours, only pin change and interrupt settings to other pin)
Even if sender somehow blocked, why not on every board.

ralcaide, 2019-05-03:
So, it seems the problem is in the original ST driver, isn't? We need to find a solution to that asap, this fact is blocking several of our ongoing projects.
Regarding sender blockage because receiver and not on every board is necessary to take next into account:
- We are using a transceiver: Same peripheral is receiving and sending
- In our previous devices if they were receiving they are not able to send. Maybe here is happening something similar.
- Is it possible somehow to stop listening and then send the answer?
- Regarding your comment in ST, it seems the code is trying to send the answer but the HW itself is not sending it. Maybe it is related with a configuration option of the S2LP. Could you take a look to its documentation
- Could you try to send several answers instead only one?

fpgamcu, 2019-05-03:
`Is it possible somehow to stop listening and then send the answer?`
I think not, because management of RF driver is done by contiki net layer. I will check code for this option.
`Regarding your comment in ST, it seems the code is trying to send the answer but the HW itself is not sending it. Maybe it is related with a configuration option of the S2LP. Could you take a look to its documentation`
Will check it.
`Could you try to send several answers instead only one?`
Already done, same result.

ralcaide, 2019-05-06:
@fpgamcu , any news regarding this problem?. I see that we got not answer in the forum. Any additional idea?. As you know, we should find a solution asap

ralcaide, 2019-05-06:
@fpgamcu , do you think the chirp status could help us to characterize the problem?. If the problem is the relation between reception-transmission then should help us.

In other hand, I think the problem is maybe related with Contiki, isn't? Could you ask also in COntiki forums?

ralcaide, 2019-05-09:
First preliminary test results looks very promising. We are right now communicating with 6 ENs through 1 ER. We will perform deeper tests and will let you know how is going.

Thanks @fpgamcu !

ralcaide, 2019-06-26:
TESTED OK in 19.24.02 https://sinapseenergia.testrail.io/index.php?/runs/view/74&group_by=cases:section_id&group_order=asc

closed 2019-06-26 by ralcaide