Sinapse-Energia/AP-Sinapse #213: [CMC Context][BUG] CMC context is not working

ralcaide, 2019-04-16:
The context CMC in version 19.12.01 is NOT working. It is not able to connect in GPRS nor in ETH nor in WIFI.

The contexts AP and EN are working OK with all the connection options.

ralcaide, 2019-04-16:
Make:

`# CMC
TOPIC=CMC_GENERIC
CONTEXT='CMC'
################################################
#CMC GPRS
INFO='CMC_GENERIC_GPRS'
CONNECTION='GPRS'

make clean
make change_cmc
make change_gprs
make config_gprs APN=$APN USR= PASSWD=
make config_mqtt_broker MQTTURI=soporte-tecnico.bitnamiapp.com MQTTPORT=1883 MQTTUSER= MQTTPASS=
make config_mqtt_topic MQTTTOPIC=$TOPIC
make config_version FWVER=$VERSION
make config_info BINFO=$INFO  
make all

cp $ORIG_BIN $DEST_PATH$DEST_BIN
CRC=$(crc32 $DEST_PATH$DEST_BIN)
echo CRC32 is $CRC
perl -e 'print pack "H*", "'$CRC'"' >> $DEST_PATH$DEST_BIN

mv $DEST_PATH$DEST_BIN $DEST_PATH$VERSION'/CL-'$CONTEXT'-'$TOPIC'-'$CONNECTION'-'$APN'-'$VERSION'.bin'`

ralcaide, 2019-06-03:
@fpgamcu , are you able to reproduce this problem? We need to to have a CMC context working

fpgamcu, 2019-06-03:
I will check it today

ralcaide, 2019-06-26:
@fpgamcu , have you checked the context CMC is working in last version (19.26.00)?

fpgamcu, 2019-06-27:
I will test and if needed fix it today

fpgamcu, 2019-07-08:
Tested on my device - latest commit is fixing issue. Now all contexts should work ok.

ralcaide, 2019-07-17:
Tested OK in 19.26.03

closed 2019-07-17 by ralcaide