Sinapse-Energia/AP-Sinapse #214: [HW][Update] Modifications required due to HW revision: EN-SA_v2p0

ralcaide, 2019-04-22:
In the new HW version several HW modifications have been included. These modifications provokes for example that the Air Quality sensors related with PMXX are not starting because now is necessary to enable the VDD / VCC. 
It is necessary to take into account the changes explained in red in the attached document, specially (high priority) the changes related with the sensors power resumed in the next table:

ON/OFF VDD1    GPIO    GPIO    PD3
ON/OFF VCC2    GPIO    GPIO    PD7
ON/OFF VDD3    GPIO    GPIO    PD14
ON/OFF VCC4    GPIO    GPIO    PD15

[[HW-FW]EN-SA-v2p0_rev1p0.pdf](https://github.com/Sinapse-Energia/AP-Sinapse/files/3103283/HW-FW.EN-SA-v2p0_rev1p0.pdf)

ralcaide, 2019-05-22:
It seems all the problems of ER in relation with its stability after restarts are related with this new version of the HW. See issue #215.

Droid has an old  version of the HW
Sinapse test with a new version of the HW

@fpgamcu, could you take a look to the differences and perform the required modifications?. The new HW baseboard is now flying to Ucrania

ralcaide, 2019-06-10:
@fpgamcu , could you please answer my previous question? Is necessary any modification in the FW?

fpgamcu, 2019-06-11:
Please check also this document, it is pins diff according to PDF you shared. Green is pins that was fixed.
https://docs.google.com/spreadsheets/d/1pyc525AS2Up_VurnDIO8D-EidRm15IX_ad9_-HyWrdA/edit?usp=sharing

ralcaide, 2019-06-11:
Thanks, we will check.
@francisjjp, can you check everything is correct?

sinapseenergia, 2019-06-11:
Yes, it is ok.

ralcaide, 2019-06-11:
Preliminary testing:

**GPRS**

BL 19.24.00 + CL 19.24.00 

- Connects OK
- 5 restart 398; : OK
- 5 restarts physically : OK

**ETH**

BL 19.24.00 + CL 19.24.00 

- Connects OK
- 5 restart 398; : OK
- 5 restarts physically : KO . After first physical restart, the LED remains always OFF

ralcaide, 2019-06-26:
TESTED OK https://sinapseenergia.testrail.io/index.php?/runs/view/73&group_by=cases:section_id&group_order=asc with version 19.24.00

closed 2019-06-26 by ralcaide