Sinapse-Energia/AP-Sinapse #215: [BUG][ETH/WIFI] The AP restarts unexpectedly and stays OFF

ralcaide, 2019-04-29:
AP with context EN and BL 1.46.0:

1.46.1: Everything works OK
19.12.01: Device connects OK and works OK during X minutes (between 5 and 30) setting up several EPDs. Then, unexpectedly the device turn OFF. Sometimes, after several minutes connects again. In other cases, the device does not connect until it is restarted.

Then we setup again the 1.46.1 and works OK again.

ralcaide, 2019-04-29:
Makefile:

################################################
#EN ETH
# TESTED OK
INFO='EN_GENERIC_ETH'
CONNECTION='ETH'
CONNECTION_COMPLEMENT='DHCP' 

make clean
make change_en
make change_wlan
make config_wlan_mode_dhcp
make config_wlan_ftc_enable
make config_mqtt_broker MQTTURI=soporte-tecnico.bitnamiapp.com MQTTPORT=1883 MQTTUSER= MQTTPASS=
make config_mqtt_topic MQTTTOPIC=$TOPIC
make config_version FWVER=$VERSION
make config_info BINFO=$INFO  
make all

cp $ORIG_BIN $DEST_PATH$DEST_BIN
CRC=$(crc32 $DEST_PATH$DEST_BIN)
echo CRC32 is $CRC
perl -e 'print pack "H*", "'$CRC'"' >> $DEST_PATH$DEST_BIN

mv $DEST_PATH$DEST_BIN $DEST_PATH$VERSION'/CL-'$CONTEXT'-'$TOPIC'-'$CONNECTION'-'$CONNECTION_COMPLEMENT'-'$VERSION'.bin'

################################################
#EN WIFI

INFO='EN_GENERIC_WIFI'
CONNECTION='TODO'
CONNECTION_COMPLEMENT='TODO' 

make clean
make change_en
make change_wlan
make config_wlan_mode_dhcp
make config_wlan_ftc_disable
make config_mqtt_broker MQTTURI=soporte-tecnico.bitnamiapp.com MQTTPORT=1883 MQTTUSER= MQTTPASS=
make config_mqtt_topic MQTTTOPIC=$TOPIC
make config_version FWVER=$VERSION
make config_info BINFO=$INFO  
make all

cp $ORIG_BIN $DEST_PATH$DEST_BIN
CRC=$(crc32 $DEST_PATH$DEST_BIN)
echo CRC32 is $CRC
perl -e 'print pack "H*", "'$CRC'"' >> $DEST_PATH$DEST_BIN

mv $DEST_PATH$DEST_BIN $DEST_PATH$VERSION'/CL-'$CONTEXT'-'$TOPIC'-'$CONNECTION'-'$CONNECTION_COMPLEMENT'-'$VERSION'.bin'

ralcaide, 2019-04-29:
With context AP is not tested. It will be done if required

ralcaide, 2019-04-29:
@fpgamcu , as you know, it is very important to fix this bug asap.

soporteHW, 2019-04-29:
Context AP:
-TODO : failed
-GPRS: ok

Context EN:
-TODO: failed
-DHCP: failed

Context CMC:
-GPRS: ok

ralcaide, 2019-05-06:
@fpgamcu , could you please provide feedback about that? Are you able to reproduce the problem? Estimation about fixing?

ralcaide, 2019-06-26:
It seems this problem was related with HW modification in the EN V2p0 and EN V3p0. Tested here https://sinapseenergia.testrail.io/index.php?/runs/view/74&group_by=cases:section_id&group_order=asc was OK with HW EN V3p0 (19.24.02)

closed 2019-06-26 by ralcaide