Sinapse-Energia/AP-Sinapse #217: [Read RS485 Generic] Read the RS485 periodically

ralcaide, 2019-05-23:
The goal of this task is to develop a command to be able to listen the interface RS485 (without modbus protocol on top of that). 
Here is explained the generic command open to modifications after the developers feedback:

[RS485 - Generic - Periodic](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#read-rs485---generic-periodic)

The main goal is of this command is to be able to read an RFID UHF reader that sends a message with the id of a tag when is detected.

The configuration of the sensor is: 57600, None, 8,1

ralcaide, 2019-05-23:
@fpgamcu , do you think the generic modbus command should work?

ralcaide, 2019-05-27:
We need something like a RS485

fpgamcu, 2019-05-27:
Listening RS485 is the same as listening UART, but need also set RS485 write control pin into read position.
It is easy to implement this feature. As I understand, we should gather received by UART bytes for PERIOD duration and each PERIOD send over MQTT message with this received bytes and timestamp of last received byte. One question - if nothing received over period, should we send empty CONTENT, or previous CONTENT and previous TIMESTAMP, or don't send anything?

fpgamcu, 2019-05-27:
In other hand generic modbus will not work because modbus have protocol layer specific to it.

ralcaide, 2019-06-26:
Thanks for the feedback. This task is not necessary by the moment

ralcaide, 2019-07-01:
This task is activated again. We will need a generic RS485 command, on demand and periodic as explained in the description. We will use this feature to read a RS485 weather station. This task should be done after fix the current CMC problems

ralcaide, 2019-07-01:
This command should work with context EN and CMC

ralcaide, 2019-07-08:
We want to use the attached device using the generic RS485 interface. The priority right now is high. @fpgamcu , could you take a look a provide feedback, please?

[CP communication protocol updated - En 201809.pdf](https://github.com/Sinapse-Energia/AP-Sinapse/files/3368110/CP.communication.protocol.updated.-.En.201809.pdf)

[RK900-05 Modbus Protocol.pdf](https://github.com/Sinapse-Energia/AP-Sinapse/files/3368111/RK900-05.Modbus.Protocol.pdf)

http://www.duolink.com.ar/wp-content/uploads/RK900-05.pdf

ralcaide, 2019-07-12:
Thanks, we will test it right now. Could you provide us please an example about a received frame and its meaning regarding the weather station?

ralcaide, 2019-07-25:
Tested by Jose https://sinapseenergia.testrail.io/index.php?/runs/view/81&group_by=cases:section_id&group_order=asc

closed 2019-07-25 by ralcaide