Sinapse-Energia/AP-Sinapse #218: [AP][ALL] Integrate 6LP features into the main branch

ralcaide, 2019-06-11:
Now, we have two parallel branches and we miss several important features in each one.
It is necessary to integrate the 6LP RF as a communication southbound in the main branch in order to have all the developed features for CMC, EN and AP accessible

fpgamcu, 2019-06-25:
By now Makefile compilation is not working for merged code, I'll fix it tonight

ralcaide, 2019-06-25:
Thanks. 
I see only two possible RFs in Definitions.h. Is that right or should be three (TDA have two differents modes)?
We wait to to makefile

fpgamcu, 2019-06-26:
Old/New RF driver mode is switched in runtime through commands,
RF/6LP mode is switched through define, USE_6LP_TRANSCEIVER enable or disable 6LP instead of RF
It is also possible to have 6LP switch in runtime, but as I checked, we have not enough RAM in order to use this feature.

ralcaide, 2019-06-26:
Thank you, everything is clear. By the moment it is OK for us like that

ralcaide, 2019-06-26:
Estimated time for the makefile?

fpgamcu, 2019-06-26:
Makefile operation was changed in latest commit, now all parameters is set directly through compiler defines.
Please use ‘make help’ in build folder for details and example.

ralcaide, 2019-06-27:
Thanks, we will test it asap.
Question: What BL should we use? Should we use a different BL depending the context or the M4_EN1.1_F427VG_V19.24.00 will work for every board?

ralcaide, 2019-07-01:
@fpgamcu , could you please answer my previous question? What BLs should be use depending the HW?

fpgamcu, 2019-07-03:
You can use BL V19.24.00

ralcaide, 2019-07-17:
Tested OK in 19.26.01

closed 2019-07-17 by ralcaide