Sinapse-Energia/AP-Sinapse #219: [CMC][Metering] L3_READ_METERING_P is not working

ralcaide, 2019-06-26:
The command L3_READ_METERING_P is not working at all in 19.12.01. To fix it

ralcaide, 2019-06-26:
@fpgamcu , have you tested this command is working properly in the last version 19.26.00 ?

fpgamcu, 2019-06-27:
I will test and if needed fix it today

ralcaide, 2019-07-08:
@fpgamcu , how is going this task? Is it fixed with last commit too?

fpgamcu, 2019-07-08:
I will be able to test with analyzer at 19:00 because Analyzer is now in other office, Roman has used it.

fpgamcu, 2019-07-09:
Here is one issue. I noticed that for ANL subspace and for EPD subspace context call GenericReadElement and GenericWriteElement with the same index for different subspaces. It will lead to erasing EPD data with ANL data and vice versa. I will study this behavior more, but for now CMC and AP context is not compatible.

ralcaide, 2019-07-09:
ok, by the moment we will use only the context EN and CMC. It is to say, we will not sue context AP. We will test the context CMC and create a new task to review the found problem

ralcaide, 2019-07-17:
Tested OK in 19.26.03

closed 2019-07-17 by ralcaide