Sinapse-Energia/AP-Sinapse #221: [6LoWPAN][Networking] To study and implement improvements in the networking protocol

ralcaide, 2019-06-28:
In the Sinapse RF protocol we've implemented some mechanisms in order to improve the RF communications like frames & bursts, repeaters, group addresses, ttl...
In 6LoWPAN we think should be already implemented some mechanisms to manage the performance of the network.

It is necessary to study different options provided by Contiki and then implement the best ones.

ralcaide, 2019-06-28:
@fpgamcu , how is working now the information sending in 6LP? Exists frames and bursts or is only sent one frame each time is required?

fpgamcu, 2019-07-03:
Now it is sending one frame (UDP message) each time. Bursts is not used now. Group addresses is possible through boadcasting and should be added, I think we have task for it. TTL is managed somewhere inside 6LP and contiki libraries. Repeaters are already implemented as 6LP is a mesh network, and in mesh network each node is in same time repeater, that translates messages further in the mesh, it is managed by 6LP and contiki libraries.