Sinapse-Energia/AP-Sinapse #222: [BUG][5;] The command 5; does not working after restarting the device

ralcaide, 2019-07-12:
We send the command 5; to connect to a new broker. It works OK but after restart with 398; it connects again to original broker. Sometimes connects to the new one but most of the times connects to the old one. The new broker is online and accepting connections

Tested with 19.26.01. 

Please fix it

fpgamcu, 2019-07-18:
Tested with 5;mine.sensorslice.com;1883;;; and 5;soporte-tecnico.bitnamiapp.com;1883;;;

ralcaide, 2019-07-25:
Tested passed in V 19.28.00: 
CMC: https://sinapseenergia.testrail.io/index.php?/runs/view/83&group_by=cases:section_id&group_order=asc

ER:
https://sinapseenergia.testrail.io/index.php?/runs/view/84&group_by=cases:section_id&group_order=asc

closed 2019-07-25 by ralcaide